chmorgan/libesphttpd #4: Example not working on ESP32 with current ESP-IDF

PaulFreund, 2017-11-03:
Hi, thank you so much for maintaining this library! It'll save me a ton of work.

I noticed that the init functions for the flash and tcpip adapter are not called. Without those no socket can be allocated and the wifi will not work.

    tcpip_adapter_init();    
    nvs_flash_init();

Is this new to ESP-IDF?

chmorgan, 2017-11-03:
HI @PaulFreund. If I'm not misunderstanding you those things are up to the user of libesphttpd to handle. Consider that there are a range of tcpip settings and wifi configuration that could occur and that is all outside of the range of an application level server like an httpd server.

If you'd like to see an example of how it is used you can look at this repository that I forked from Sprite and updated to work with the latest libesphttpd and esp-idf (esp32).

https://github.com/chmorgan/esphttpd-freertos

PaulFreund, 2017-11-03:
Oh man I am sorry this issue was meant for the esphttpd-freertos repository! I needed to add those two in the app_main function in user_main.c.

chmorgan, 2017-11-03:
I see them here:

https://github.com/chmorgan/esphttpd-freertos/blob/master/main/user_main.c#L230

Have you pulled to make sure you have the latest version?

PaulFreund, 2017-11-03:
Oh I overlooked that but the nvs_flash_init is missing completely, without it the esp will reboot because of errors in the wifi stack. 

I just tested without tcpip_adapter_init in main and I get "platHttpServerTask: failed to create sock!" again. I think the task is not fast enough to be run before httpdInit.

chmorgan, 2017-11-03:
Yeah that code still kind of sucks and I could see that it could be racy. Why is it using a separate thread in any case? If you'd like to remove the separate thread and instead call it as a function that would be a great pull request.

PaulFreund, 2017-11-03:
If I'm already do a pull reqeust should I change the example to AP mode so no access data has to be supplied?

chmorgan, 2017-11-03:
Sure. I've merged your PR in and am closing this issue now. Thank you again for the pull request!

closed 2017-11-03 by chmorgan