chmorgan/libesphttpd #7: printf() vs. httpd_printf() used inconsistently in some files

dmcnaugh, 2017-11-06:
`printf()` has sometimes been used rather than` httpd_printf()` in (but not limited to) the following files:
* httpd-freetos.c
* httpd.c
* espfs.c

this causes unwanted messages to be output even when `httpd_printf()` has silenced.

P.S. what's the best practice for  silencing `httpd_printf()`?

chmorgan, 2017-11-07:
Imo it is probably ok to change these all to use httpd_printf(). Then in another change you could add a compile time conditional to platform.h to compile out the httpd_printf() calls, like:

#ifdef ENABLE_PRINTING
...
#else
  #define httpd_printf(fmt, ...) (void)0
#endif

See https://stackoverflow.com/questions/5765175/macro-to-turn-off-printf-statements

Pull request welcome!

chmorgan, 2018-01-20:
Closing this ticket out as httpd_printf() has been transitioned to use the esp log component.

closed 2018-01-20 by chmorgan