chmorgan/libesphttpd #8: Problem with ESP32 example

marbalon, 2017-11-08:
I've just tried to run esphttpd-freertos example of libesphttpd on ESP32. First I go to menuconfig to select esphttpd and uart port etc. Compilation works fine, and server starts without any problems, I can connect to AP, and load default webpage with cats ;)

But, first page loading slow about 20-30seconds, and if I tried to run tests, all tests filed. When I tried to load led.tpl it's  page is not responding.

Maybe I miss something in menuconfig ?

valkuc, 2017-11-08:
Which browser do you use?

marbalon, 2017-11-08:
Chrome.

valkuc, 2017-11-08:
Could you please try in IE and tell me page load speed.

marbalon, 2017-11-08:
Loadins speed is similar, and one time some of tests passed, but then hang. And now after reset I can't pass any test in IE of course.

On the console I have an output:
`URL = /test/
Is url index 0
Is url index 15
Pool slot 0 is done. Cleaning up for next req
Sel add 4097 (write 1)
Sel add 4098 (write 0)
Sel add 4099 (write 0)
Sel add 4100 (write 0)
Sel add listen 4096
sel ret
Sel add 4097 (write 0)
Sel add 4098 (write 0)
Sel add 4099 (write 0)
Sel add 4100 (write 0)
Sel add listen 4096
sel ret
Pool slot 1: socket closed.
Pool slot 2: socket closed.
Pool slot 3: socket closed.
Sel add 4097 (write 0)
Sel add listen 4096
sel ret
Pool slot 0: socket closed.
Sel add listen 4096`

valkuc, 2017-11-08:
Well then looks like you faced something different than me. I have a problem that pages in IE loading smooth and fast, while in other browsers (Chrome, FF) load speed is very slow. But this is on ESP8266. Have not run it on ESP32 at all.

chmorgan, 2017-11-08:
@marbalon I assume you are running without ssl support enabled? If so page loading should be very quick.

marbalon, 2017-11-08:
Could you upload sdkconfig for this example ?

chmorgan, 2017-11-08:
@marbalon here is the one that I tested with before pushing the changes out to git.

https://ufile.io/01j4n

marbalon, 2017-11-08:
Thanks, but your config works as mine ... :(

It looks like sometimes it just hangs without any error/reboot on conolse. When I run the test few test passed rapidly and then hangs and the rest of the tests failed.

chmorgan, 2017-11-08:
Maybe the tests are identifying something that isn't working correctly
inside of libesphttpd. I had no idea the tests existed until you mentioned
them!

I can try running them at some point to see what occurs but I have a big
backlog of other work on a separate project.

If you find anything I'd appreciate info on what it might be!

marbalon, 2017-11-08:
Hmm I changed wifi mode form AP to STA and everything seems to works fine... except wifi template - Loading but doesn't shouw anything except searching, but this is minor problem.

I also tried to find information what is the licence for libesphttpd, can be used with closed source project ???

chmorgan, 2017-11-08:
My fork of libesphttpd is MPL licensed. So you can use it with a closed source project but if you redistribute it you have to redistribute any changes/improvements you've made to libesphttpd.

Sprite wrote the original libesphttpd and it has a Beer-ware license that is effectively public domain. Imo it would be better to use my fork as it is significantly more developed and currently maintained.

marbalon, 2017-11-08:
I first found Sprite version, but I found it is quite old, then I checked forks and yours is the most active fork and works with current esp-idf! I will let you know when I found solution why it is not working with AP mode. But you made your tests in AP or STA mode ?

chmorgan, 2017-11-08:
@marbalon in my implementation it is working very well in both AP and STA and APSTA mode.

I haven't run these tests here that you are seeing fail. It's a good catch that they exist and I'll run them here at some point to see if they fail.

I'm guessing you'll have more experience with libesphttpd with this failing case in the near term but if you do see what might be going wrong I can help or review changes.

marbalon, 2017-11-08:
OK, will let you know, now I need to find solution to compile this lib with my existing sources. 

I put libesphttpd in to component folder, select it in menuconfig but got error:

CC build/libesphttpd/util/esp32_flash.o
CC build/libesphttpd/util/captdns.o
CC build/libesphttpd/util/cgiwifi.o
make[1]: *** No rule to make target '/home/marcin/esp/RFIDC1/spiffs', needed by 'webpages.espfs'.  Stop.

Folder with html files also exists... I need to browse makefiles to find solution.

chmorgan, 2017-11-08:
Which component folder? spiffs is part of esp-idf, it is found under esp-idf/components/spiffs.

In my case the directory structure is:

/someproject
/someproject/esp-idf (esp-idf project)
/someproject/app (Makefile and sdkconfig are here)
/someproject/app/components/
/someproject/app/components/libesphttpd
/someproject/app/main/main.c (app/main/ is where the project source files are)
/someproject/app/html (html files here)

marbalon, 2017-11-08:
Yes, exactly the same. But I tried with clean example + this lib and works fine, should be something related with my project settings.

marbalon, 2017-11-08:
Ok, found. I've changed html to spiffs dir in my configuration... sorry for stupid problem.

chmorgan, 2017-11-08:
Glad to see it is working now!

marbalon, 2017-11-08:
I saw that cgi for OTA operations are not prepared for ESP32, I tried them but got system reset. Did you tried it on the ESP32 ?

chmorgan, 2017-11-08:
Yep. They aren't supported yet. I have some in-progress work that is
working for esp32 OTA but haven't pushed it out yet. Coming soonish!

marbalon, 2017-11-08:
Good to know, can't wait to test it ;-) 
I'm glad that is ready to use http server for esp-idf because it is missing part in this SDK.

chmorgan, 2017-11-08:
There is nghttp in esp-idf, isn't that a web server as well?

marbalon, 2017-11-08:
Yes, but without so many nice features like this have.. ;-)

Btw. I found cgiwifi.c is also not ready for ESP32...

marbalon, 2017-11-10:
Problem resolved. It was my (another) stupid mistake I tested ESP on my desk, but have router next to the ESP with channel 1....  like in example project.

closed 2017-11-10 by marbalon