chmorgan/libesphttpd #12: Contribute back to the community

morganrallen, 2017-11-22:
"The MPLv2 restricts freedom in that it requires modifications to be given back to the community. This license establishes the agreement that in exchange for using this great library that users are required to give back their changes to let others benefit."

So how about pushing back upstream already?

chmorgan, 2017-11-22:
Hello again!

Do you mean pushing them to Sprite's repo? I could potentially push them to Sprite's but he wasn't interested in relicensing to mpl. It would seem a bit passive-aggressive to open a PR against his repository given that he'd have to do that in order to incorporate them. Certainly I wouldn't be opposed to that if he were interested.

valkuc, 2017-11-22:
The original upstream license is beer-ware. Furthermore, Sprite does not maintain his repo last time.

morganrallen, 2017-11-22:
You can push changes sans license changes. Just depends were you priorities are, actually providing utility to the community or licensing semantics.

chmorgan, 2017-11-22:
@morganrallen not to re-hash it again but I prefer copyleft licenses. Providing my changes back without a copyleft license isn't interesting to me.

morganrallen, 2017-11-22:
despite the mass hypocrisy of pushing for a license with the sole purpose of FORCING people to contribute back while you refuse to do the same for the ORIGINAL author?

chmorgan, 2017-11-22:
@morganrallen I wasn't interested in putting in few hundred hours of effort without a copyleft license. License changed and a ton of improvements have been made that can be used by anyone that shares back their changes per that license. I followed the original license and even got contributor permissions and I'm sharing everything here in public.

morganrallen, 2017-11-22:
"Yeah, I gave it some thought and decided against it." -- Spritetm
I also did not give you permission. In fact, I layed out several concerns.

* The present license is in no way prohibiting contributions. -- which you keep insisting is the case
* Forking just to relicense WILL cause fragmentation (are you going to still push to Sprite/libesphttpd?) -- you've succeeded there!

You then went on to explain your motivations are due to your working on a commercial product. So at the very least, correct the blatant lie at the end of the readme and explain your true motivations for relicensing.

chmorgan, 2017-11-22:
@morganrallen we'll have to agree to disagree.

closed 2017-11-22 by chmorgan

chmorgan, 2017-11-22:
@morganrallen and I’ll update the readme with more details of why I pushed for re-licensing. I’m not hiding that it was because of a ton of work on the library for a commercial project.