chmorgan/libesphttpd #14: cmo branch: unexpected behavoir when open a second web session to the same url

ataweg, 2018-02-05:
Hi Chris,
when I open a second web session to the https server the browser hangs, and the ESP32 disconnects from the Wifi. But it doesn't crash. Since this didn't happened in an older version of libesphttpd, I have the suspicion that this could be a problem in the last version. Stackoverflows were not reported.
Do you have an idea in which corner I should investigate further?
Viele Gruesse
Axel

https://github.com/ataweg/libesphttpd/tree/awe

chmorgan, 2018-02-06:
Hi @ataweg are you using 51084e? I found and fixed an issue with a rewrite that was causing issues.

Are you seeing any console/terminal prints? Depending on the number of connections you've given the server it may be that you are using them up.

Many of the rewrites on this branch are an attempt to find a latent heap corruption issue that I can't pin down to either the server or a relatively large amount of other code. So I started rewriting anything that was suspect and found and fixed a number of potential issues without fail.

If you can confirm that you are using that version then you have the latest fixes to this branch. In my testing it works well, I can hit cgi endpoints thousands of times with 8 simultaneous connections before triggering a fault.

If you have codespace you should enable, via make menuconfig, stack smashing protection overall, and heap corruption detection comprehensive. I'm interested in what results you get. If you see the same heap issues then that helps to rule out the other code I have here but doesn't rule out an issue in some other esp32 library.

What version of esp-idf are you using?

ataweg, 2018-02-06:
Hello Chris,
I changes to your original code 51084e and got the same behavior .I used the latest IDF version  532107c .  To compile it successfully, I had to fix a few corners, see attached patch. 

I had also im my own version the heap corruption enabled.  Currently I use CONFIG_ESPHTTPD_STACK_SIZE=12288 and all other definition are at their default values from the configuration. Do you have some recommendation for the stack and heap sizes?

Since my old version works, I will try to bring both versions together somewhere in the middle.

[libesphttpd_51084e.patch.txt](https://github.com/chmorgan/libesphttpd/files/1700329/libesphttpd_51084e.patch.txt)

closed 2018-02-06 by ataweg

ataweg, 2018-02-06:
Hello Chris, 
I have it.  I reduced the maxConnections from 16 to 8.  I the first case there is a free heap size of 12K bytes and this seems to be not enough. Now there is around 40K bytes free heap and it works now as with my older version of libesphttp.
Thank you for your support.
Viele Gruesse
Axel

chmorgan, 2018-02-06:
Ahh. Yeah if you run other parts of the esp32 systems out of heap they may
block. Such as WiFi reception. I’m guessing of course but I know WiFi does
dynamic memory allocation, if you haven’t turned that off it might be the
case where the system is wedged waiting for free heap so it can continue.

That seems like a very very large stack. I think mine is 4K. Any reason why
yours was so large?