chmorgan/libesphttpd #15: no zlib.h

fastworks, 2018-03-23:
I've installed it as directed into the current esp-idf library. This is using a windows install as directed in the expressif instructions. esp-idf examples and such build and work correctly so I don't suspect it is the toolchain per se.

/home/miker/esp/esp-idf/components/libesphttpd/espfs/mkespfsimage//main.c:30:18: fatal error: zlib.h: No such file or directory

chmorgan, 2018-03-23:
Looks like you need the zlib development package installed. What linux
distro?

fastworks, 2018-03-23:
It is not a linux distro at all, it's using the windows toolchain. I think the best solution may be to embed a zlib copy in the package, provided it is license-compatible.

chmorgan, 2018-03-24:
Hmm. I’m not a frequent windows user. It could be possible to add zlib as a
submodule. How do other projects handle build dependencies on windows?

valkuc, 2018-03-24:
Compile under cygwin.

chmorgan, 2018-03-24:
Hey, long time!

Should we add a note to the readme.md about building under windows?

fastworks, 2018-03-25:
I'm not sure how dependencies are typically handled in windows as I'm more of a unix user. My project requires some windows based tools unfortunately. I think the best strategy for compatibility would be to include zlib as part of this component. I'm pretty sure there are non-GNU infected versions of zlib that could be included without conflicting with your license.

edclombardo, 2018-04-22:
How does one initialize the http webserver from chmorgan/libesphttpd source code? httpdInit() is not found in the source code, where it is found in Spritetm/libesphttpd source code for ESP8266.  I am looking at using libesphttpd for ESP32 from chmorgan's git repository.  I found httpdFreertosInit() but it requires arguments that appear to me to be internal to the libesphttpd and not known in app_main().  Almost like httpdInit() is missing.  Any suggestions would be appreciated.  Also how to handle zlib on windows+eclipse.

wdouglas1999, 2018-05-23:
Good day,
I am on Windows and have the same issue as fastworks. The issue of “no zlib” 
Is there any work around available for this ?
Thank you.

phatpaul, 2018-07-20:
I figured it out.
In the esp-idf (mingw32.exe) install zlib-devel

```
$ pacman -S zlib-devel
```

commit 1e77f7ff (2019-03-06): Add a comment. resolves #15
closed 2019-03-06 by phatpaul