chmorgan/libesphttpd #18: Cannot flash example....

wdouglas1999, 2018-05-23:
Good Day.
Thank you for the great code you have written. I am having an issue with flashing example code on to my esp32.
Any assistance will be greatly appreciated.
The following error pops up:
LD build/user.elf
C:/msys32/home/wdoug/esp/esphttpd-freertos/main/build/esp32\libesp32.a(cpu_start.o):(.literal.main       _task+0x20): undefined reference to `app_main'
C:/msys32/home/wdoug/esp/esphttpd-freertos/main/build/esp32\libesp32.a(cpu_start.o): In function `       main_task':
C:/msys32/home/wdoug/esp/esp-idf/components/esp32/cpu_start.c:452: undefined reference to `app_mai       n'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/msys32/home/wdoug/esp/esp-idf/make/project.mk:388: /home/wdoug/esp/esphttpd-freertos       /main/build/user.elf] Error 1

tidklaas, 2018-05-24:
This repository is not a standalone application, it is just a library to be included in an application. For a simple example try https://github.com/chmorgan/esphttpd-freertos instead.

wdouglas1999, 2018-05-24:
Thank you for the reply. I have managed to include the library, but now I have the error of “no zlib” Connot be found.
I am using Windows.
Do you have any idea on how to solve this?

tidklaas, 2018-05-24:
Solve the Windows problem? Sure, just install Linux...

wdouglas1999, 2018-05-24:
Hahahah! Funny thing you mention. I actually installed it last night. Just setting up eclipse now.
Thx

wdouglas1999, 2018-05-24:
I set USE_GZIP_COMPRESSION to “no” and compiles and flashes with no issues.
Thx

wdouglas1999, 2018-05-24:
Is there any plan to add support to scan and connect the esp32 to wifi network. I notice the example does not support this.

chmorgan, 2018-05-25:
I think I just merged a pull request in this week that improves support for
the wifi interaction. I'll gladly merge in PRs that implement such
functionality if it isn't present.

wdouglas1999, 2018-05-25:
I tried to configure the ESP32 using the example  to connect to my WIFI network, but it does not seem to work. 
I click on “Configure your WIFI network “ .
Then it says scanning but no networks are displayed. If I click scan in STA + AP mode the esp32 reboots but still no networks are displayed.
Can you please advise on what I am doing wrong?
Could this be because I am not using zlib to compress?

tidklaas, 2018-05-27:
Hmm, at first I could not reproduce the reboot issue. After updating the esp-idf, I get a stack overflow in the WiFi event task when the scan done call-back gets executed:
```
[0;32mI (57197) cgiwifi: wifiStartScan[0m
[0;32mI (60037) cgiwifi: wifiScanDoneCb[0m
***ERROR*** A stack overflow in task eventTask has been detected.
abort() was called at PC 0x4008daac on core 0

Backtrace: 0x4008d8bc:0x3ffd9f50 0x4008da93:0x3ffd9f70 0x4008daac:0x3ffd9f90 0x4008b348:0x3ffd9fb0 0x4008cac4:0x3ffd9fd0 0x4008ca7a:0xa5a5a5a5

Rebooting...
```
Increasing the stack size to 3k solves the problem:
`CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE=3072`

wdouglas1999, 2018-05-27:
I confirm it is working now. Thank you for your effort.

closed 2018-08-03 by chmorgan