chmorgan/libesphttpd #21: Support for ESP-IDF feature/cmake

PaulFreund, 2018-06-11:
Hi,

do you already plan to support he new CMake based toolchain? The current CMakeLists.txt is unfortunately not compatible. Depending on your timeline it is possible I support but I can't promise anything at the moment.

Best regards,

Paul

chmorgan, 2018-06-12:
Hi @PaulFreund.

I'm not aware of this new cmake toolchain. Do you have any details or links to info about it?

PaulFreund, 2018-06-12:
Yeah sure,

Here is the announcement: https://esp32.com/viewtopic.php?f=13&t=5559&p=24109
Here are the docs: https://esp-idf.readthedocs.io/en/feature-cmake/get-started/windows-setup.html
And here is the branch itself: https://github.com/espressif/esp-idf/tree/feature/cmake

chmorgan, 2018-06-12:
Ahh now that makes more sense.

I've been a fan and a user of cmake for a number of years now. The esp-idf
build system does work impressively well due to their simple and logical
conventions. I can't say I'd be itching to replace it given how well it
works but clearly someone else is doing the work there...

I'd be fine with supporting the new cmake stuff. It looks like its windows
only for now? Maybe if I start using it under Mac I'll add support but
until then I think I'll have to rely on people like yourself that have an
itch and a need for the improvements. I can certainly help provide some
guidance if someone wants to develop it, I'm quite familiar with cmake.

Si03, 2018-08-02:
I have no idea how to set up for the ESPIDF CMake but would love to use it. Here is what I get in a clean install and setup as per the ESP CMAKE documentation:
```
C:\Users\Simon\esp\lm_ycm_bridge>c:\python27\python C:\Users\Simon\esp\esp-idf\tools\idf.py build
Setting IDF_PATH environment variable: C:\Users\Simon\esp\esp-idf
Running cmake in directory C:\Users\Simon\esp\lm_ycm_bridge\build
Executing "cmake -G Ninja --warn-uninitialized C:\Users\Simon\esp\lm_ycm_bridge"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.17.1.windows.1")
-- Performing Test flag_supported
CMake Error at C:/Program Files/CMake/share/cmake-3.11/Modules/CheckCSourceCompiles.cmake:98 (try_compile):
  try_compile command is not scriptable
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.11/Modules/CheckCCompilerFlag.cmake:50 (CHECK_C_SOURCE_COMPILES)
  C:/Users/Simon/esp/lm_ycm_bridge/components/libesphttpd/CMakeLists.txt:8 (check_c_compiler_flag)
  C:/Users/Simon/esp/lm_ycm_bridge/components/libesphttpd/CMakeLists.txt:15 (enable_c_compiler_flag_if_supported)
  scripts/expand_requirements.cmake:141 (include)
  scripts/expand_requirements.cmake:173 (expand_component_requirements)

CMake Error at C:/Users/Simon/esp/esp-idf/tools/cmake/project.cmake:64 (include):
  include could not find load file:

    C:/Users/Simon/esp/lm_ycm_bridge/build/component_depends.cmake
Call Stack (most recent call first):
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1

```

I'll keep googling but if anyone has any pointers then great!
Si.

N-Coder, 2018-11-19:
@chmorgan it seems that Mac and Linux are now also [supported](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system-cmake.html), at least I'm using it under Fedora together with CLion right now. :)
Also, the "CMake-based build system will become the default build system in ESP-IDF V4.0. The existing GNU Make based build system will be deprecated in ESP-IDF V5.0."

To use it together with libesphttpd, I tried the following steps:

- I cloned this repo as a submodule to `components/libesphttpd`,
- added `set(COMPONENT_REQUIRES libesphttpd)` right before `register_component()` in `main/CMakeLists.txt`.
- and added `set(EXTRA_COMPONENT_DIRS components)` right before `include($ENV{IDF_PATH}/tools/cmake/project.cmake)` in the main `CMakeLists.txt`.

Unfortunately, as this project's `CMakeLists.txt` doesn't comply with the new 'standard', the build will still fail due to various `command is not scriptable` errors. There is a [conversion script](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system-cmake.html#migrating-from-esp-idf-gnu-make-system) from the GNU make system, but I couldn't get it to run with this project. It seems that someone would have to rewrite the `CMakeLists.txt` according to the new definitions to make this include-able as a component via cmake.

Elektrik1, 2019-02-04:
So basically, current CMake toolchain is not supported with this component ? As it errors like that:

> -- Performing Test flag_supported
> CMake Error at C:/Program Files/CMake/share/cmake-3.11/Modules/CheckCSourceCompiles.cmake:98 (try_compile):
>   try_compile command is not scriptable
> Call Stack (most recent call first):
>   C:/Program Files/CMake/share/cmake-3.11/Modules/CheckCCompilerFlag.cmake:50 (CHECK_C_SOURCE_COMPILES)
>   D:/msys32/home/esp-idf/components/libesphttpd/CMakeLists.txt:8 (check_c_compiler_flag)
>   D:/msys32/home/esp-idf/components/libesphttpd/CMakeLists.txt:15 (enable_c_compiler_flag_if_supported)
>   D:/msys32/home/esp-idf/tools/cmake/scripts/expand_requirements.cmake:107 (include)
>   D:/msys32/home/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
>   D:/msys32/home/esp-idf/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)
> 
> 
> CMake Error at D:/msys32/home/esp-idf/CMakeLists.txt:39 (message):
>   Failed to expand component requirements

chmorgan, 2019-02-04:
@Elektrik1 correct, there is no cmake support currently except for use on Linux for testing, profiling etc.

Pull requests are very welcome for cmake support!

jkent, 2019-10-12:
Note: I've worked on this with PR #77 ... I'd love to hear feedback on it!

closed 2020-10-30 by phatpaul