chmorgan/libesphttpd #23: Additional cgi functions available "as is" for ESP32 VFS static file serving and file uploading

dmcnaugh, 2018-06-14:
Hi, I have developed a module to provide two additional cgi functions that I have named
- cgiEspVfsHook
- cgiEspVfsUpload
as the names suggest and the title indicates, these provide for serving static files from an ESP32 VFS, typically FAT FS mounted from an attached SD card, and for HTTP file uploads to the ESP32 VFS.

The are used, for example, like:
    {"/store", cgiEspVfsUpload, "/sdcard/www/store", NULL},
    {"/public/*", cgiEspVfsHook, "/sdcard", NULL},
    {"*", cgiEspVfsHook, "/sdcard/www", NULL},

Note:
- the VFS filesystem needs to already be mounted and available, that isn't done in these modules
- the static files can be manually gzipped on the filesystem and will be matched and served with the correct encoding header without having to include the .gz extension in the URL
- code for supporting templates is included, but has never been tested. Its modelled off an older version of httpdespfs.c 
- some things might be specific to my requirements (like the returned content at the completion of a file upload)

They work for me, so I am unlikely to put any more effort into maintaining them or improving them. I am happy to provide them "as is" for you or others to review/rework and better generalise into libesphttpd if anyone finds them useful.

Is you want me to provide the code please tell me the best way to send it. I could do it as a pull-request, or set it up as a gist, or anything else you want to suggest.

chmorgan, 2018-06-14:
Wow very cool man. A pull request would be great. I’ve thought about such a
cgi plug-in after using the vfs a bit. Imo this may be a better approach
than the current packed and compiled in case that libesphttp supports but
I’m guessing it depends on use case. Building the pages in is far simpler
than vfs I think if you are using a separate flash partition. Maybe vfs can
use a compiled in variable as well?

chmorgan, 2018-06-17:
Merged, closing!

closed 2018-06-17 by chmorgan