chmorgan/libesphttpd #30: how to remove a double-slash in url

phatpaul, 2018-07-23:
Some modern webservers will automatically ignore a double-slash
in the URL.  (i.e. http://192.168.1.79//min.json is equivalent to http://192.168.1.79/min.json)
(see https://webmasters.stackexchange.com/questions/8354/what-does-the-double-slash-mean-in-urls )

How can I have the esphttpd server ignore the double-slash?

chmorgan, 2018-07-24:
It should be possible to update the libesphttpd code to ignore the repeated slashes but I can't say I know where to look at updating that. Maybe someone else using the library can help out? Or you can submit a PR if you figure it out.

phatpaul, 2018-07-24:
Thanks Tido!
That did the trick.  Now I'm able to interface with an industrial device that (incorrectly?) requests files with multiple slashes.
I guess you should get to work now ;-)

chmorgan, 2018-08-03:
@phatpaul @tidklaas's change has been merged into master, closing this issue.

closed 2018-08-03 by chmorgan