chmorgan/libesphttpd #33: does libesphttpd work asynchronously?

squonk11, 2018-08-06:
Currently I am searching for a webserver for ESP32. There are different solutions out there - each with specific features and problems. With your solution I like that it is based on ESP-IDF and that it supports websockets and SSL. 
But does it also allow real parallel access from several clients being seved simultaneously? In other words: does it work asynchronously?

chmorgan, 2018-08-06:
Hello.

There is a single processing task that handles each connection as well as
new connections. Sockets are handled as they are activated with new data so
you will be able to have several clients communicating at the same time but
the cpu would be jumping between processing each one of them.

It may be possible to fork them off but the esp32 has limited cpu power and
memory. I wonder if it would be of a significant benefit to handle a task
per connection. It should certainly be possible to modify the core code to
do so if you wanted to try it out.

Can you say more about your use case? Also esp-idf has started adding their
own web server implementation. I’m not sure on the details of that
implementation though.

Chris

squonk11, 2018-08-06:
Sorry, I did not phrase my question correctly. My use case is the following: I am planning to set up a webserver for an ESP32 which is connected via a serial interface to an industrial device. In this way the parameterization of this device shall be possible using a browser. This device has roughly 3000 parameters which need to be read by a browser via the webserver within short time (few seconds). The webpage shall be programmed in html/javascript using a framework like react or vue.js. The additional difficulty here is that the webserver must communicate via https (SSL/TLS). In such a scenario the browser first reads the index.html and then all other files referenced in index.html (css, js, jpg, ...). This is typically then done with several separate https requests. Also reading the 3000 parameters is done with 3000 separate https requests. Unfortunately each https request takes quite a long time to be served due to the TLS handshake which takes 1-2 seconds. Due to this the loading of the page might take a longer time and most problematic: reading out all 3000 parameters takes many minutes and not few seconds. 
So, that is why I was thinking about using a persistent  connection (http header: "connection: keep-alive"). But when I do this the ESP32 can only serve one client at a time because the browser will not close this connection; and if the connection is not closed the single thread for serving this connection cannot serve the connection of another client.
That is why I was asking for a real parallel access from several clients (using separate threads for each connection).
I hope this explanation gives a clearer picture of what I am planning to do.

chmorgan, 2018-08-06:
HI @squonk11 some of this makes sense but the few seconds per request doesn't make as much.

You can configure any number of maximum clients to the libesphttpd server, typically depending on how much memory you have available. The SSL connections are expensive, memory-wise, like 16k per connection.

Your web browser should be smart enough to connection pool, Chrome certainly is, so requests to the ssl server should take a second to establish initially due to the key exchange, but then should be very fast.

So you should be ok with four or five max clients. If there is an issue with reusing a connection that is something that should be fixed on the server. From my testing the first request is slow but follow up requests are very quick, using https of course.

Have you tested tried it out? It should only take a few minutes to set up a https server on the esp32 development board if you start with the libesphttpd esp-idf example.

squonk11, 2018-08-07:
Hi @chmorgan 
until now I tested the HttpServer class of [this](https://github.com/nkolban/esp32-snippets) repository. Unfortunately the https functionaltity which is implemented using mbedtls did not work for me. So I implemented in those sources encryption using openssl. This finally worked. But the big disadvantage was, that for each https request there was the key exchange because each connection was closed after the response was transmitted. This finally made the https conversation very slow, whereas http was fast enough. Then I tried to implement into kloban's sources the handling of a persistent connection (header: "connection: keep-alive") firing up a separate task for each request. But with this I failed due to unknown reasons - the webserver always crashes after a short time. 
Now I am searching for another solution and found your libesphttpd and [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer). The latter seems to be quite complete but unfortunately is based on Arduino - I would like to work using esp-idf. 
I think I will set up a test using your lib during the next days (unfortunately I currently do not have much time for this) and check if here I also have the problem with key exchange for each https request. I will let you know about the result.
kind regards
   Lothar

chmorgan, 2018-08-07:
Hi @squonk11. Just tested here with postman and 14.255s for the initial request, 242ms for subsequent requests.

squonk11, 2018-08-07:
Hi, that looks a little bit strange to me - especially the 14 seconds fo the initial request is much longer than what I had during my tests using the HttpServer class (including https) from Kolban. Also the 242ms for the subsequent requests is still quite long. For my 3000 parameters this would take roughly 12 minutes. Thas is still too long. I was expecting 5 to 10ms for each parameter. Some time ago I made a test using mongoose but http only (no https) - here the target was reached.
But nevertheless I will have a try with your library during the next days.

chmorgan, 2018-08-07:
There is also an mdns name lookup occurring in my case so it could be other things causing more overhead.

The call I was making was hitting the filesystem on the device as well as parsing certificates. A lower overhead call is between 36ms and 150ms consistently.

There may be more that can be done on the libesphttpd side as well to reduce overhead. The esp32 isn't the fastest processor and the aes encryption is going to add some overhead to each exchange. Http is much faster.

On the implementation side, is there any reason why you can't send more parameters at once, or group them up and report them as a larger chunk? I mean you could probably send all 3000 parameters as a single response pretty quickly. Do you not have control over the code that is reading the parameters?

squonk11, 2018-08-07:
yes, you are right. I was already also thinking about optimizations on the protocol to the external device. Although I can not change that protocol but I could use the ESP32 as some kind of proxy or cache. For this I would need some storage space; ESP32s RAM is quite "expensive". Cacheing the data on an SD card could be an option here. Another possible optimization would be using a websocket connection for reading the parameters, since a websocket connection is persistent by design. So, basically here I see chances for optimizations. 
But the loading of the webpage probably will still take too much time. The application will be a single page application using react or vue.js. The page will consist of several (maybe 20) files; if every file takes 14 seconds for transmission, that would be far too slow. 
This all needs to be checked before I finally decide for a webserver library.

chmorgan, 2018-08-07:
Keep in mind you can cheaply get an esp32-wroom with 4mb of spiram. I use
the one analog-lamb builds, alb32-wroom.

chmorgan, 2019-09-16:
Closing due to inactivity.

closed 2019-09-16 by chmorgan