chmorgan/libesphttpd #36: WebSockets?

klapligehesten, 2018-08-29:
Hi,

I am currently playing around with WebSockets in the newest build of the libesphttpd server.

I seems to run fine except from sending continuous frames  frames. I have tried out a lot of different ways to get it working and have given up for now, but i will be happy if someone can tell me if there is something wrong with my code.

The following code is sending the frames ok but chrome i keeps saying:
Error: WebSocket connection failed: Received start of new message but previous message is unfinished.

```
......
void ws_send_frames( WebSocket *m, 	P_NODE p) {
	P_BME_LOG_REC b;
	char message[100];
	
	sprintf( message, "{\"bme280_log\":");
	
	cgiWebsocketSend( &httpdFreertosInstance.httpdInstance, ws, message, strlen( message), WEBSOCK_FLAG_MORE);
	
	P_NODE t = p;
	while(t != NULL) {
		b = (P_BME_LOG_REC)t->data;
		
		sprintf( buffer, "{\"bme280\":{\"ts\":\"%llu\",\"t\":\"%0.1f\",\"p\":\"%0.3f\",\"h\":\"%0.2f\"}}",
				b->timestamp, b->humidity, b->pressure, b->temperature);
				
		cgiWebsocketSend( &httpdFreertosInstance.httpdInstance, ws, message, strlen( message), WEBSOCK_FLAG_NONE);
		
		t = t->next;
	}
	
	cgiWebsocketSend( &httpdFreertosInstance.httpdInstance, ws, message, strlen( message), WEBSOCK_FLAG_NONE);
	
}
......
```
Cheers - HPSchultz

chmorgan, 2018-08-30:
HI @klapligehesten.

I'm not sure if the web socket implementation in libesphttpd is working correctly. Have you looked at the echo example? I think it may be in the demo repository.

If you do end up figuring out what is going on I'd appreciate info here or a pull request.

Chris

klapligehesten, 2018-08-30:
Hi Chris,
Thank you for the answer.
Yes, i have looked at he examples i could find. Including the echo example written by Spritetm.
Judged by your answer, it seems no one has fully tested out the 'continuation frame' scenario.
I really like the libesphttpd server implementation and the current release is running very stable :-)
I think its worth the work, so I will take a look at the RFC 6455 WebSocket Protocol document and the libesphttpd implementation.

I'll get back as soon i have something worked out 
Cheers - HPSchultz

PaulFreund, 2019-04-17:
Maybe we can run the autobahn suite against this implementation to find out where it has shortcomings https://github.com/crossbario/autobahn-testsuite