chmorgan/libesphttpd #38: multipartBoundary pointer not correct

PaulFreund, 2018-09-11:
https://github.com/chmorgan/libesphttpd/blob/9634e28f6197da0af41d9b67e9c1c9e526d1d3ec/core/httpd.c#L734

I found there was a change in commit 21fb7796d1d4e0a514416b4d7198d79c3e2a654b where you removed two character overrides with '-'. I am a little confused about both states. 

We are searching for "boundary=" ->b and later set the pointer to b+7. The length of our search subject is 9 so our buffer will contain for example "y=----WebKitFormBoundarywrAqgDbLLFmgN4Zn". Is there a reason the y= have previously been overwritten with dashes? Should we exchange the 7 for strlen("boundary=")?

chmorgan, 2018-09-13:
@PaulFreund that change was intended to avoid modifying memory. Where is that multipartBoundary value even used in the code? I don't see why we should have to modify the memory at all but I can't figure out where the code that needs that data modified exists or why multipartBoundary doesn't appear used.

PaulFreund, 2018-09-13:
I fully agree that modifying memory is the wrong way. I think we have two ways to continue here. 
Either we change the length to "b+9" and have the correct start of the boundary value, 
or we create a buffer for the boundary and copy it inside together with the two following hyphens that are needed.
I'm using the multipart boundary to upload OTA updates to MicroSD and would be okay to create my own buffer inside this function including the hyphens so I'm in favor for b+9.
What do you think?

PaulFreund, 2018-09-25:
Sorry I was working on another project. I choose to correct the boundary pointer and not change memory / prepend the hyphens. This will break existing uses of the multipart boundary but as it is broken already this will be a "real" solution. As we are memory constrained I don't think we should hold a copy of the boundary with hyphens for everybody even if they are not using it (most likely 90% of users)

PR #40 (2018-09-25): Fix issue #38, invalid multipartBoundary
This commit fixes #38
closed 2018-09-25 by chmorgan