chmorgan/libesphttpd #39: shrink_decoder.c: No such file or directory

versamodule, 2018-09-24:
I place the libesphttpd repository into the components directory of your esp-idf folder.
ran menuconfig and enabled it in menuconfig. 
I then copied the esphttpd-freertos, and ran make on it. I get the following error. 

C:/msys32/home/JIm/esp/esp-idf/components/libesphttpd-master/espfs/heatshrink_decoder.c:18:50: fatal error: ../lib/heatshrink/heatshrink_decoder.c: No such file or directory compilation terminated.

There are no files in the /lib folder.

mharizanov, 2018-09-24:
Check the installation instructions, you need to init the git submodules as well

versamodule, 2018-09-24:
Thank you
I did that and now I get this. 
/home/JIm/esp/Projects/esphttpd-freertos/components/libesphttpd/espfs/mkespfsimage//main.c:30:18: fatal error: zlib.h: No such file or directory
 #include <zlib.h>

mharizanov, 2018-09-24:
sudo apt-get install zlib1g-dev

versamodule, 2018-09-24:
Again, thank you. I am running a Windows PC using mingw

phatpaul, 2019-03-06:
See #15

closed 2019-03-06 by phatpaul