chmorgan/libesphttpd #42: OTA max bin filesize?

phatpaul, 2018-10-04:
I'm testing the OTA upgrade function cgiUploadFirmware.  My .bin file size is ~1.3Meg.  
The OTA partition size is 4Meg.  (16Meg total flash size)

The OTA process starts and seems to repeat mid-way.  Then the ESP32 reboots without any warning message.

Then it appears to be running the new OTA image after rebooting, but I'm afraid it didn't complete flashing it.

Is there a size limit to the HTTP POST data?  Is there a watchdog timer expiring?
Shouldn't there be some console message indicating that the process completed before reboot?

```
I (521305) ota: Running partition type 0 subtype 16 (offset 0x00420000)
checkBinHeader: e9 4008 3f400020
I (521305) ota: Writing to partition subtype 17 at offset 0x830000
I (521305) ota: esp_ota_begin succeeded
E (521305) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521305) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521305) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521305) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521315) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521315) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521315) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521315) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521315) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521315) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521315) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
E (521315) wifi: esf_buf: t=8 l=1600 max:16, alloc:16 no eb
I (523585) esp_image: segment 0: paddr=0x00830020 vaddr=0x3f400020 size=0x855c0
(546240) map
I (523795) esp_image: segment 1: paddr=0x008b55e8 vaddr=0x3ffb0000 size=0x04ca8
( 19624)
I (523805) esp_image: segment 2: paddr=0x008ba298 vaddr=0x3ffb4ca8 size=0x00000
(     0)
I (523805) esp_image: segment 3: paddr=0x008ba2a0 vaddr=0x40080000 size=0x00400
(  1024)
0x40080000: _iram_start at C:/msys32/home/labview/To
ols/esp-idf/components/freertos/xtensa_vectors.S:1685

I (523805) esp_image: segment 4: paddr=0x008ba6a8 vaddr=0x40080400 size=0x05968
( 22888)
I (523825) esp_image: segment 5: paddr=0x008c0018 vaddr=0x400d0018 size=0x7f6cc
(521932) map
0x400d0018: _stext at ??:?

I (524015) esp_image: segment 6: paddr=0x0093f6ec vaddr=0x40085d68 size=0x0f7f8
( 63480)
0x40085d68: ieee80211_output_process at ??:?

I (524035) esp_image: segment 7: paddr=0x0094eeec vaddr=0x400c0000 size=0x00000
(     0)
I (524035) esp_image: segment 8: paddr=0x0094eef4 vaddr=0x50000000 size=0x00000
(     0)
I (524035) esp_image: segment 0: paddr=0x00830020 vaddr=0x3f400020 size=0x855c0
(546240) map
I (524245) esp_image: segment 1: paddr=0x008b55e8 vaddr=0x3ffb0000 size=0x04ca8
( 19624)
I (524255) esp_image: segment 2: paddr=0x008ba298 vaddr=0x3ffb4ca8 size=0x00000
(     0)
I (524255) esp_image: segment 3: paddr=0x008ba2a0 vaddr=0x40080000 size=0x00400
(  1024)

I (524255) esp_image: segment 4: paddr=0x008ba6a8 vaddr=0x40080400 size=0x05968
( 22888)
I (524265) esp_image: segment 5: paddr=0x008c0018 vaddr=0x400d0018 size=0x7f6cc
(521932) map
0x400d0018: _stext at ??:?

I (524465) esp_image: segment 6: paddr=0x0093f6ec vaddr=0x40085d68 size=0x0f7f8
( 63480)
0x40085d68: ieee80211_output_process at ??:?

I (524495) esp_image: segment 7: paddr=0x0094eeec vaddr=0x400c0000 size=0x00000
(     0)
I (524495) esp_image: segment 8: paddr=0x0094eef4 vaddr=0x50000000 size=0x00000
(     0)
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)

```

Here's my parition table:
```
# -------------------------------------------------------
# -    Partition layout generated by BUILD.sh script    -
# -------------------------------------------------------
# Name,         Type, SubType, Offset,  Size,       Flags
# -------------------------------------------------------
nvs,            data, nvs,     0x9000,  16K,
otadata,        data, ota,     0xd000,  8K,
phy_init,       data, phy,     0xf000,  4K,
MicroPython,    app,  factory, 0x10000, 4160K,
MicroPython_1,  app,  ota_0,   ,        4160K,
MicroPython_2,  app,  ota_1,   ,        4160K,
internalfs,     data, fat,     ,        3840K,
```

I thought that these options might be incorrect, but it seems that most of them are not even used for ESP32? 

```
#define OTA_FLASH_SIZE_K 4096
#define OTA_TAGNAME "generic" // not used in ESP32

CgiUploadFlashDef uploadParams={
	.type=CGIFLASH_TYPE_FW,
	.fw1Pos=0x1000,
	.fw2Pos=((OTA_FLASH_SIZE_K*1024)/2)+0x1000, // not used in ESP32
	.fwSize=((OTA_FLASH_SIZE_K*1024)/2)-0x1000, // not used in ESP32
	.tagName=OTA_TAGNAME
};
```

chmorgan, 2018-10-04:
Hi @phatpaul I'm using Ota update here with about a 1.2MB image without any issues.

It does sound like you are running into the watchdog issue.

Check your sdkconfig, I think I had to extend some wtd.

```
CONFIG_INT_WDT=y
CONFIG_INT_WDT_TIMEOUT_MS=1000
CONFIG_INT_WDT_CHECK_CPU1=y
CONFIG_TASK_WDT=y
CONFIG_TASK_WDT_PANIC=
CONFIG_TASK_WDT_TIMEOUT_S=5
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0=y
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=y
```

phatpaul, 2018-10-09:
It looks like it was working.

I was expecting the console to show "Upload done. Sending response"  But that was a debug message and I only had Info set.

Perhaps change that console output to Info level, since the begin message is also Info.

I (521305) ota: esp_ota_begin succeeded

closed 2018-10-09 by phatpaul

phatpaul, 2018-10-11:
I'm making some additions/improvements to the cgiFlash API and improving the html GUI.
Would you be interested in merging it in?  I'm not bothering to support the older ESP8266 chip (I don't have one).

The API is documented here:
https://github.com/phatpaul/libesphttpd/blob/master/README-flash_api.md

Attached a screenshot of the GUI.
[image]

chmorgan, 2018-10-11:
Hi Paul.

Sure, that would be great. I think someone with an esp8266 could always back port those changes. I think the 8266 is still pretty popular but the esp32 is certainly the largest growth and where things are going. As long as the changes don’t trash entirely the 8266 support I’ll merge them in.

Btw have you had a look at the esp-idf web server implementation? I know the espressif guys are working on it but haven’t checked up on it.

Chris