chmorgan/libesphttpd #44: cgiEspVfsUpload() crashes at strncat()

phatpaul, 2018-10-16:
I'm trying to upload a file via HTTP PUT using the supplied cgi function cgiEspVfsUpload().

(file downloading via the accompanying function cgiEspVfsHook() is working)

I added upload handler:
```
HttpdBuiltInUrl builtInUrls[]={
...
	ROUTE_CGI_ARG("/upload/*", cgiEspVfsUpload, FS_BASE_PATH "/html"), // FATfs filesystem
	ROUTE_CGI_ARG("*", cgiEspVfsHook, FS_BASE_PATH "/html"), // FATfs filesystem
	ROUTE_FILESYSTEM(), // espFs filesystem

	ROUTE_END()
};
```
And I test upload with cURL:  
``` 
$ curl -X PUT -T test.txt "http://192.168.33.173/upload/test.txt"
curl: (56) Recv failure: Connection reset by peer
```
The ESP then crashes ```Guru Meditation Error: Core  1 panic'ed (LoadProhibited) at pc=40099419. Setting bp and returning..```

I trapped it with debugger.  It broke at strncat() at strncat.c:78 0x40099419	
 (I don't have the source code for that function.  Where can I find it?)

The calling function in the stack is cgiEspVfsUpload() at esp32_httpd_vfs.c:243 0x40138589

[image]

Some clues:
* Seems cgiEspVfsUpload() wants to append string connData->getArgs to filename, but that string is empty!  -- but it doesn't seem that should cause a crash?!
* Seems to me the correct string to append would be connData->url, since that is where the rest of the file path is specified.

phatpaul, 2018-10-16:
Seems this stuff was never tested?
Anyway, I'm overhauling it. 

I fixed the PUT handler in e2ace1cbc95fbcbb2f6146ffcbb4f1a02dadc01c .  

Next I will work on a POST handler (seems POST is more commonly used on the web for file uploads).

I also am working on a html/js GUI example for file uploading to filesystem.

phatpaul, 2018-10-17:
@chmorgan that function is to handle a PUT, not a GET. 
It is my understanding that a PUT typically doesn't include any arguments.  (a POST does)

Examples:

- download a file with GET http://1.2.3.4/filetoget.txt 
  - no arguements
- invoke cgi with GET http://1.2.3.4/api?arg1=1&arg2=2
- upload file via PUT http://1.2.3.4/filetoupload.txt
  - no arguments (filename is in the URL)
- upload file via POST http://1.2.3.4/upload.cgi
  - arguments are in the header and body (filename is not in the URL)
```
Content-Type: multipart/form-data; boundary=---------------------------25661900231176
Content-Length: 3704
-----------------------------25661900231176
Content-Disposition: form-data; name="file"; filename=".x.png"
Content-Type: image/png
... file contents here...
-----------------------------25661900231176--
```

So I'm writing a POST handler for uploading files, since most all of the example client code I can find (ajax file managers etc.) use POST, not PUT for file upload.

phatpaul, 2018-10-17:
I'm replacing the function cgiEspVfsUpload() with 2 separate functions:
```
//This is a HTTP PUT handler for uploading files to the VFS filesystem.  Put is one of 2 options to upload files -- see POST upload handler as well.
//
// Specify base directory as 1st cgiArg.
// Usage:
//      ROUTE_CGI_ARG("*", cgiEspVfsPut, "/base/directory/")
CgiStatus cgiEspVfsPut(HttpdConnData *connData);

//This is a HTTP POST handler for uploading files to the VFS filesystem.  POST is one of 2 options to upload files -- see PUT upload handler as well.
//
// Specify base directory or single file as 1st cgiArg.
// Usage:
//      ROUTE_CGI_ARG("/filesystem/upload.cgi", cgiEspVfsPost, "/base/directory/") 
//           - Allows creating/replacing files anywhere under "/base/directory/".
//      ROUTE_CGI_ARG("/writeable_file.txt", cgiEspVfsPost, "/base/directory/writeable_file.txt") 
//           - Allows only replacing content of one file at "/base/directory/writeable_file.txt".
CgiStatus cgiEspVfsPost(HttpdConnData *connData);
```
And here is an example route table (just the end):
```
HttpdBuiltInUrl builtInUrls[]={
...
	ROUTE_REDIRECT("/altui/filesystem/", "/altui/filesystem/index.html"),
	// FATfs filesystem POST handler. Allows only replacing content of one file at "/spiflash/html/writeable_file.txt".
	ROUTE_CGI_ARG("/writeable_file.txt", cgiEspVfsPost, FS_BASE_PATH "/html/writeable_file.txt"), 
	
	// FATfs filesystem POST handler.  Allows creating/replacing files anywhere under "/spiflash/html/".  (note trailing slash)
	ROUTE_CGI_ARG("/filesystem/upload.cgi", cgiEspVfsPost, FS_BASE_PATH "/html/"), 
	
	// FATfs filesystem PUT handler, will fall through if http req is not PUT.
	ROUTE_CGI_ARG("*", cgiEspVfsPut, FS_BASE_PATH "/html"), 
	
	// FATfs filesystem GET handler, will fall through if file not found.
	ROUTE_CGI_ARG("*", cgiEspVfsGet, FS_BASE_PATH "/html"), 
	
	// espFs filesystem GET handler.
	ROUTE_FILESYSTEM(), 

	ROUTE_END()
};
```
Notice that the POST version allows some constraints on which files can be changed.  

And I've created a GUI for uploading files with either POST or PUT:
[image]

I'll send a PR after I've tested it some more.

chmorgan, 2018-10-17:
@phatpaul I think this make sense. You are also correct on the handler, I misread the code to be == PUT not != PUT. I'll look for the pr, thanks for working on this stuff.