chmorgan/libesphttpd #50: Libesphttpd, SSL and ESP8266

roger-zacharczyk, 2018-11-23:
Hello,

Have you tested SSL on ESP8266 as well as on ESP32? Does it work, considering SSL's memory/cpu needs for, say, 2-10 simultaneous clients with reasonable connect/read timeouts set?

chmorgan, 2018-11-23:
Hello.

I developed the ssl support on esp32 and I’m pretty confident it is mostly correct, hard to say it’s perfect of course.

Haven’t tried on esp8266.

I think it’s something like 30k per ssl connection, I think I added those details in the code somewhere or the readme, so you’d need a lot of memory for 10 connections but it should be possible with esp32 if you have one with psram.

Chris

chmorgan, 2019-09-16:
Closing due to inactivity.

closed 2019-09-16 by chmorgan