chmorgan/libesphttpd #52: How to properly shot down esphttpd?

mharizanov, 2018-12-19:
Folks,

I need to only temporarily start esphttpd, and once it is done - I need to properly shut it down, freeing all allocated resources. 

What will be the proper way to do it with the existing code?

Regards

chmorgan, 2018-12-19:
Hi @mharizanov.

If you look in include/httpd.h, there is:

#ifdef CONFIG_ESPHTTPD_SHUTDOWN_SUPPORT
void httpdShutdown(HttpdInstance *pInstance);
#endif

So if you enable shutdown support you can call this function to shut the server down.

mharizanov, 2018-12-20:
I see, thanks! I'll give it a try later today.

mharizanov, 2018-12-20:
Managed to get it working, more details here: https://github.com/chmorgan/esphttpd-freertos/issues/8

I admit that I didn't do proper research prior to asking.

Thanks for your support!

closed 2018-12-20 by mharizanov