chmorgan/libesphttpd #53: cgi-wifi errors not propagated to client

phatpaul, 2019-01-03:
I identified errors that are not propagated to the client via JSON.  Will fix with a PR.

1. If try to AP scan when not in STA mode, UI does not show an error and the command is retried indefinitely.

2. If STA is trying to connect (but some error is preventing it from connecting), then scan fails, but UI is not aware (keeps starting scan).  This condition should be handled somehow (timeout connecting if not connected after xx s?  abort connecting as soon as scan-APs commanded?)

Unrelated change I propose: don't scan APs automatically from UI because it interrupts ongoing WiFi communications.  This is causing me a headache when one may leave the WiFi page open in another browser tab and then complain that service is intermittent.
I'm adding a warning and a "Start Scanning" and "Stop Scanning" and "Scan Once" buttons to the UI.

Another unrelated change I propose: I will extend the cgiWiFiConnStatus function to be able to give info for all interfaces (Eth, STA, AP).  URL argument can specify which interface, but default will send all interfaces.  JSON response something like : 
```
{ 
  "AP": { 
    "status": "Connected",
    "SSID": "poco",
    "encryption": "WPA2",
    "IP": "192.168.4.1",
    "MASK": "255.255.255.0",
    "GW": "0.0.0.0",
    "clients": "2"
  },
  "STA": { 
    "status": "Connected",
    "SSID": "janie2",
    "encryption": "WPA2",
    "IP": "192.168.2.108",
    "MASK": "255.255.255.0",
    "GW": "102.168.2.1"
  },
  "Eth": { 
    "status": "Connected",
    "IP": "192.168.2.102",
    "MASK": "255.255.255.0",
    "GW": "102.168.2.1"
  }
}

```
Where status could be one of 
        "Failed",
        "Connected",
        "Idle",
        "Update",
        "WPS Start",
        "WPS Active",
        "Connecting",
        "Fall Back"

tidklaas, 2019-06-01:
The main problem is that the ESP-IDF does not provide an API to retrieve the current configuration and state of the networking / WiFi stack without keeping track of the various event notifications. The CGI functions should be able to just query the current state or update the configuration in a stateless way, without having to set up event listeners and timer call-backs.

One way around this is to wrap all the configuration/scanning handling into an external component with a stateless query/update API. I wrote a kind of WiFi management module for this some time ago, but so far it is not much more than a proof of concept. If anybody is interested, I could put it up here on GitHub.

phatpaul, 2019-09-04:
PR #72 fixes number 1 above.  
Also don't scan APs automatically.

closed 2019-09-25 by phatpaul