chmorgan/libesphttpd #54: 200ms ACK delay sometimes

phatpaul, 2019-01-16:
I'm noticing some 200ms delays in transferring very small http files.  

Wireshark shows that the delay appears to happen when the server waits for an ACK from the client before sending the next frame.

The delay seems dependent on the size of the file (for example, a 1kB file takes ~220ms, but a 1.5kB file takes ~8ms)

I think it may have to do with the TCP, specifically Nagle's algorithm.  Is it possible to eliminate this delay? 
Can we set TCP_NODELAY?

FYI: 
There's tons of references to this online for example: 
https://access.redhat.com/documentation/en-US/Red_Hat_Enterprise_MRG/1.2/html/Realtime_Tuning_Guide/sect-Realtime_Tuning_Guide-Application_Tuning_and_Deployment-TCP_NODELAY_and_Small_Buffer_Writes.html

(I do see one reference to Nagle's algorithm here: https://dl.espressif.com/doc/esp-idf/latest/api-reference/kconfig.html#tcp-oversize)

chmorgan, 2019-01-17:
Hi Paul.

Is there a way to set this flag in lwip?

I think it’s fine for us to disable tcp nodelay but maybe cia a flag? I’m
not sure that always disabling it is the correct thing to do. Thoughts?

Chris

phatpaul, 2019-01-17:
Chris,
Yes I think it would be great to enable/disable this via a MENUCONFIG flag.
I'm looking into how to set it via LWIP.

I see the option TCP_NODELAY defined in lwip/sockets.h:248
```
 * Options for level IPPROTO_TCP
 */
#define TCP_NODELAY    0x01    /* don't delay send to coalesce packets */
```
and I guess I can use it just like the other options are set in httpd-freertos.c:388.

So it becomes:
```
                int keepAlive = 1; //enable keepalive
                int keepIdle = 60; //60s
                int keepInterval = 5; //5s
                int keepCount = 3; //retry times
                int nodelay = 0;
#ifdef CONFIG_ESPHTTPD_TCP_NODELAY
                nodelay = 1;  // enable TCP_NODELAY to speed-up transfers of small files.  See Nagle's Algorithm.
#endif
                setsockopt(remotefd, SOL_SOCKET, SO_KEEPALIVE, (void *)&keepAlive, sizeof(keepAlive));
                setsockopt(remotefd, IPPROTO_TCP, TCP_KEEPIDLE, (void*)&keepIdle, sizeof(keepIdle));
                setsockopt(remotefd, IPPROTO_TCP, TCP_KEEPINTVL, (void *)&keepInterval, sizeof(keepInterval));
                setsockopt(remotefd, IPPROTO_TCP, TCP_KEEPCNT, (void *)&keepCount, sizeof(keepCount));
                setsockopt(remotefd, IPPROTO_TCP, TCP_NODELAY, (void *)&nodelay, sizeof(nodelay));
```

How does this look?

phatpaul, 2019-01-17:
It looks like that did the trick.

Wow my app is feeling much faster now!

I'll post a PR shortly.

chmorgan, 2019-01-26:
Hi @phatpaul, thanks for the PR, just merged it in.

closed 2019-01-26 by chmorgan