chmorgan/libesphttpd #58: Simple http auth does not protect CGI scripts.

marbalon, 2019-03-01:
Hi,

I'm using this component on ESP32 withc basic http auth. But the problem is that usre can call cgi scripts even it is not authorized. For example my configuration is:
...
ROUTE_CGI("/admin/example.cgi", cgiExample),
ROUTE_AUTH("/admin/*", webBasicAuth),
...
And user can call cgi dirrectly when enter full url.
Is this OK or this is a BUG ? If this is OK, I think I need to verify autorization in CGI script, but is there any function/macro providing infromation that current request comes from authorized browser ?

BR,
Marcin.

phatpaul, 2019-03-06:
The route table is scanned from top-to-bottom, so your auth entry needs to be before anything that you want protected.

marbalon, 2019-03-07:
Thank you @phatpaul - now it is working as expected.

closed 2019-03-07 by phatpaul