chmorgan/libesphttpd #59: httpd limit connections

phatpaul, 2019-03-08:
So modern web-browsers will try to open 6+ simultaneous connections to the webserver.  If I have more than 2~3 clients connected to my web App, the server starts dropping connections and the browser hangs for a while.

It seems that if (MAX_CONNECTIONS in libesphttpd) is >= (LWIP_MAX_SOCKETS in menuconfig), then httpd doesn't even know that LWIP is dropping the connections.  And LWIP_MAX_SOCKETS has a max value of 16 (why?).
I propose putting a compiler warning if MAX_CONNECTIONS >= LWIP_MAX_SOCKETS.

When MAX_CONNECTIONS = 32, but LWIP_MAX_SOCKETS  = 16, I see lots of these errors when connect 4 clients:
```
E (21552) httpd-freertos: accept failed
accept: Too many open files in system
E (21562) httpd-freertos: accept failed
accept: Too many open files in system
```

So I decreased MAX_CONNECTIONS to 14.  Now when exceeded I see error in console:
```
I (2260322) httpd-freertos: all 14 connections in use on '0.0.0.0'
E (2260332) httpd: send buf tried to write 1032 bytes, wrote -1
I (2260332) httpd-freertos: listening for new connections on '0.0.0.0'
I (2260342) httpd-freertos: all 14 connections in use on '0.0.0.0'
I (2263762) httpd-freertos: listening for new connections on '0.0.0.0'
```
But the browser still hangs.

I feel there should be a more graceful way to refuse the connections.  Then the browser could use one of it's existing connections instead of hanging waiting for a response on a broken one.

Also, I would like a way to limit the number of connections per client (by IP?) to 2 or 3.  That way I can support more active clients.  (Actually I need to always allow the websocket connection, but limit the normal http to 1~2).

I found some guidance on what the server should do when it reaches it's connection limit:
https://stackoverflow.com/questions/35186725/how-to-limit-the-number-of-connections-to-a-socket-and-trigger-timeout-on-client

> as long as the listening socket is open, connections will be accepted by the OS, even if not being handled by the application. If you want to reject connections after a set number, you have basically two options:
> 
>     - simply accept and close directly after accepting.
>     - close the listening socket upon reaching the limit and reopen when done.
> 

Looking a the httpd code,
I see the message coming from https://github.com/chmorgan/libesphttpd/blob/master/core/httpd-freertos.c#L344
But I never see the close happen at https://github.com/chmorgan/libesphttpd/blob/master/core/httpd-freertos.c#L375

jkent, 2019-04-05:
I've done this in my repo to fix the issue when using the httpd with a the captive dns.
https://github.com/jkent/esp32-esphttpd/commit/cc6a5c99

Perhaps a way to configure the default flags of the httpd should be added?

phatpaul, 2019-04-25:
Thanks for the suggestion although it's not immediately clear to me how this will improve my case or what side-effects it will have on other use-cases.

I'll get back around to this testing http performance on my project after I complete some tasks...  

In the mean-time would you provide more info about what this would do?

jkent, 2019-05-30:
Sorry for getting back to you so late.  This closes the socket after every request, instead of keeping multiple idle sockets waiting on a request.  I've tested this on Android with the captdns and it solved my problems with WiFi setup not being reachable.  Something that also might help along with this "fix" is increasing the listen() backlog... I haven't looked into this yet.

chmorgan, 2019-05-31:
Are we keeping connections open based on keep alive? https://en.wikipedia.org/wiki/HTTP_persistent_connection#Keepalive_with_chunked_transfer_encoding

I don't see that in the code. Imo if we used that connection header we could be smarter about keeping connections open. Maybe we are keeping connections open even if the client isn't requesting keep-alive and that's why we are running out of connections?

jkent, 2019-05-31:
Yes, and yes.
> In HTTP 1.1, all connections are considered persistent unless declared otherwise.

There may be a better way to handle this, but this is where things happen:
https://github.com/chmorgan/libesphttpd/blob/748f589afe6ee39d7b2ff2593caf459b3cdef17f/core/httpd.c#L227-L229

chmorgan, 2019-05-31:
And we are using that connection string flag to close the connection after
sending the response? I didn’t spot that in my review although I also
didn’t spot or recall the code you attached either.

I’ve also been hit with the too many open connections issue but wasn’t able
to get to the bottom of it.

jkent, 2019-05-31:
Sorry, I didn't mean to link from my PR commit hash.  Here is the link to the code in master:

https://github.com/chmorgan/libesphttpd/blob/dc6c265b5df23b82162bb3e3c2f9a2b7c1ac8e3f/core/httpd.c#L227-L229

connStr is sent with the headers at the start of a http response.

Problem is that connections are not given the header "Connection: close" header by default because HFL_CHUNKED is set:

https://github.com/chmorgan/libesphttpd/blob/dc6c265b5df23b82162bb3e3c2f9a2b7c1ac8e3f/core/httpd.c#L679

PR #63 is a possible solution to this problem.  But perhaps having chunked encoding + "Connection: close" header would be a good idea.

chmorgan, 2019-05-31:
I don’t understand the implications of chunked. Can’t we close after the
response is finished sending or does chunked imply keep alive?

I can merge that pr, or Paul can, if you think it’s a worth while
improvement. If we resolved this issue we might not need it though.

Chris

jkent, 2019-05-31:
Recall that persistent connections are a default feature of HTTP/1.1

Chunked encoding lets you send a document in chunks that you don't know the size of until you are ready to send them.  Before each chunk is sent a chunk size is sent.  A size of 0 means the document is complete.  This is how a persistent connection knows that the response is complete with an unknown sized document.  It is up to the browser to close the connection when it is done with a socket.  With HTTP/1.0 this wasn't a problem because there were no persistent connections, documents did not need a Content-Length header.

Sprite_tm had this worked out with this function though it is not immediately clear:
https://github.com/chmorgan/libesphttpd/blob/dc6c265b5df23b82162bb3e3c2f9a2b7c1ac8e3f/core/httpd.c#L210-L221

With my PR, I'm just allowing the default mode to be set at compile time.

chmorgan, 2019-05-31:
That does sound familiar, that persistent is the default and the behavior with chunked.

Maybe we are seeing this issue because we are so limited in our connection number. Browsers will keep a few open, or open several to pull several files. I had also considered and suspected this was the case.

If you’d like me to merge in that PR let me know and I’ll pull it in. Or if you think its safe we should enable it for everyone, what would the downsides be?

jkent, 2019-05-31:
I think its safe to enable it for everyone.  The only downside is speed, with having more TCP handshakes.  I'd like to hear @phatpaul 's opinion first.  I think the httpd in idf has this same problem.

phatpaul, 2019-10-09:
I finally tested your PR #63 @jkent and it does indeed seem to solve my connection limit issue.
Without the single request option enabled, I quickly exceed the max connections with just 3 browser windows viewing my App.  (using 2 instances of firefox -noremote and one instance of Chromium)
```I (87812) httpd-freertos: all 14 connections in use on '0.0.0.0'```
Windows Resource Monitor shows a bunch of connections to my ESP32 (run perfmon /res)
[image]

With the single request option enabled, there is only one connection per browser in resource monitor --this is probably the single websocket connection used by my app, all other http connections have been closed as soon as the static content is loaded.
I can hit reload on all 3 browsers but I never see the 'all connections in use' message.  :+1: 

Now I was concerned that it might affect performance, but it seems the app loading time has only increased a negligible amount:
Without single request option, app loads in about 340ms:
[image]

With single request option, app loads in about 340ms:
[image]

Thanks!

closed 2019-10-09 by phatpaul