chmorgan/libesphttpd #60: CONFIG_ESPHTTPD_SSL_SUPPORT not detected

Kiogora, 2019-03-16:
Hi,

I followed the instructions in the readme but I notice the sdkconfig option CONFIG_ESPHTTPD_SSL_SUPPORT is not detected thus the build defaults to mode non-ssl.

I have also tried to add "sdkconfig.h" to the respective files including httpd-freertos, and also to the app_main file in the example usage repo.

Any thoughts are appreciated.

chmorgan, 2019-03-16:
Hi @Kiogora. Have you run 'make menuconfig' and enabled that flag under the libesphttpd section?

Kiogora, 2019-03-16:
Hi @chmorgan . Yes I have. Just as info, this is running under linux and esp-idf version 3.1 stable branch

chmorgan, 2019-03-17:
Well that flag you are setting via menuconfig is the CONFIG_ESPHTTPD_SSL_SUPPORT flag. Is menuconfig indicating why it isn't able to be set or are you seeing any build warnings/errors that point you in that direction? To debug further it would be helpful to see your sdkconfig file. Maybe there is another dependency missing or some changes occurred in 3.1 that broke it. @phatpaul are you building or using ssl with latest esp-idf?

phatpaul, 2019-03-17:
Sorry I have not tried building ssl or linux.

To be clear, do you mean you are using a linux environment to build for
esp32 or are you building libesphttpd for linux?

Kiogora, 2019-03-17:
@phatpaul To be clear, I am using a linux environment to build for ESP32.

@chmorgan There are no errors. Kindly see build messages below:
```
GENCONFIG
Building partitions from /home/alois/Documents/build_systems/espressif/esp-idf/components/partition_table/partitions_singleapp.csv...
CC build/app_trace/app_trace_util.o
CC build/app_update/esp_ota_ops.o
CC build/app_trace/host_file_io.o
AR build/aws_iot/libaws_iot.a
CC build/bootloader_support/src/bootloader_flash.o
CC build/app_trace/app_trace.o
CC build/bootloader_support/src/flash_qio_mode.o
AR build/app_update/libapp_update.a
CC build/bt/bt.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_trace/libapp_trace.a
CC build/bootloader_support/src/efuse.o
AR build/bt/libbt.a
CC build/coap/libcoap/src/address.o
CC build/bootloader_support/src/bootloader_common.o
CC build/coap/libcoap/src/async.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader_support/src/secure_boot.o
CC build/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/coap/libcoap/src/block.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/coap/libcoap/src/coap_time.o
CC build/coap/libcoap/src/debug.o
CC build/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader_support/src/flash_partitions.o
CC build/coap/libcoap/src/encode.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader_support/src/esp_image_format.o
CC build/coap/libcoap/src/hashkey.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/coap/libcoap/src/mem.o
CC build/coap/libcoap/src/net.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/coap/libcoap/src/option.o
CC build/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/coap/libcoap/src/pdu.o
CC build/coap/libcoap/src/resource.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
AR build/bootloader_support/libbootloader_support.a
CC build/coap/libcoap/src/str.o
CC build/console/linenoise/linenoise.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/console/argtable3/argtable3.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/coap/libcoap/src/subscribe.o
CC build/coap/libcoap/src/uri.o
CC build/coap/port/coap_io_socket.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
CXX build/cxx/cxx_guards.o
AR build/coap/libcoap.a
CC build/console/split_argv.o
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
[... 1345 lines cut ...]
#
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
```

Kiogora, 2019-03-17:
The resultant sdkconfig.h:

```
/*
 *
 * Automatically generated file; DO NOT EDIT.
 * Espressif IoT Development Framework Configuration
 *
 */
#define CONFIG_ESP32_PHY_MAX_TX_POWER 20
#define CONFIG_TRACEMEM_RESERVE_DRAM 0x0
#define CONFIG_FREERTOS_MAX_TASK_NAME_LEN 16
#define CONFIG_FATFS_LFN_NONE 1
#define CONFIG_TCP_RECVMBOX_SIZE 6
#define CONFIG_FATFS_CODEPAGE_437 1
#define CONFIG_TCP_WND_DEFAULT 5744
#define CONFIG_PARTITION_TABLE_OFFSET 0x8000
#define CONFIG_SPIFFS_USE_MAGIC_LENGTH 1
#define CONFIG_ESPTOOLPY_FLASHSIZE_4MB 1
#define CONFIG_IPC_TASK_STACK_SIZE 1024
#define CONFIG_FATFS_PER_FILE_CACHE 1
#define CONFIG_ESPTOOLPY_FLASHFREQ "40m"
#define CONFIG_MBEDTLS_KEY_EXCHANGE_RSA 1
#define CONFIG_UDP_RECVMBOX_SIZE 6
#define CONFIG_FREERTOS_QUEUE_REGISTRY_SIZE 0
#define CONFIG_MBEDTLS_AES_C 1
#define CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED 1
#define CONFIG_ESP32_WIFI_SOFTAP_BEACON_MAX_LEN 752
#define CONFIG_MBEDTLS_GCM_C 1
#define CONFIG_ESPTOOLPY_FLASHSIZE "4MB"
#define CONFIG_HEAP_POISONING_DISABLED 1
#define CONFIG_SPIFFS_CACHE_WR 1
#define CONFIG_BROWNOUT_DET_LVL_SEL_0 1
#define CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER 1
#define CONFIG_SPIFFS_CACHE 1
#define CONFIG_INT_WDT 1
#define CONFIG_MBEDTLS_SSL_PROTO_TLS1 1
#define CONFIG_ESP_GRATUITOUS_ARP 1
#define CONFIG_MBEDTLS_ECDSA_C 1
#define CONFIG_ESPTOOLPY_FLASHFREQ_40M 1
#define CONFIG_LOG_BOOTLOADER_LEVEL_INFO 1
#define CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE 0
#define CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS 1
#define CONFIG_MBEDTLS_ECDH_C 1
#define CONFIG_MBEDTLS_KEY_EXCHANGE_ELLIPTIC_CURVE 1
#define CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM 10
#define CONFIG_MBEDTLS_SSL_ALPN 1
#define CONFIG_ESPHTTPD_PROC_PRI 4
#define CONFIG_MBEDTLS_PEM_WRITE_C 1
#define CONFIG_BT_RESERVE_DRAM 0x0
#define CONFIG_FATFS_FS_LOCK 0
#define CONFIG_IP_LOST_TIMER_INTERVAL 120
#define CONFIG_SPIFFS_META_LENGTH 4
#define CONFIG_ESP32_PANIC_PRINT_REBOOT 1
#define CONFIG_MBEDTLS_ECP_DP_BP384R1_ENABLED 1
#define CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED 1
#define CONFIG_CONSOLE_UART_BAUDRATE 115200
#define CONFIG_LWIP_MAX_SOCKETS 10
#define CONFIG_LWIP_NETIF_LOOPBACK 1
#define CONFIG_EMAC_TASK_PRIORITY 20
#define CONFIG_TIMER_TASK_STACK_DEPTH 3072
#define CONFIG_TCP_MSS 1436
#define CONFIG_MBEDTLS_ECP_DP_CURVE25519_ENABLED 1
#define CONFIG_FATFS_CODEPAGE 437
#define CONFIG_ESP32_DEFAULT_CPU_FREQ_160 1
#define CONFIG_ULP_COPROC_RESERVE_MEM 0
#define CONFIG_LWIP_MAX_UDP_PCBS 16
#define CONFIG_ESPTOOLPY_BAUD 921600
#define CONFIG_INT_WDT_CHECK_CPU1 1
#define CONFIG_ADC_CAL_LUT_ENABLE 1
[... 138 lines cut ...]
#define CONFIG_FREERTOS_ISR_STACKSIZE 1536
#define CONFIG_OPENSSL_ASSERT_DO_NOTHING 1
#define CONFIG_WL_SECTOR_SIZE_4096 1
#define CONFIG_OPTIMIZATION_LEVEL_DEBUG 1
#define CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED 1
#define CONFIG_MBEDTLS_ECP_DP_SECP192R1_ENABLED 1
#define CONFIG_ESPHTTPD_SSL_SUPPORT 1
#define CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED 1
#define CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_ECDSA 1
#define CONFIG_SYSTEM_EVENT_QUEUE_SIZE 32
#define CONFIG_ESP32_WIFI_TX_BUFFER_TYPE 1
#define CONFIG_ESPTOOLPY_BAUD_921600B 1
#define CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED 1
#define CONFIG_LWIP_LOOPBACK_MAX_PBUFS 8
#define CONFIG_SPI_FLASH_ROM_DRIVER_PATCH 1
#define CONFIG_SPIFFS_PAGE_SIZE 256
#define CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED 1
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 1
#define CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT 3072
#define CONFIG_MONITOR_BAUD_OTHER_VAL 115200
#define CONFIG_NEWLIB_STDOUT_LINE_ENDING_CRLF 1
#define CONFIG_ESPTOOLPY_PORT "/dev/ttyUSB0"
#define CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS 1
#define CONFIG_ESP32_WIFI_IRAM_OPT 1
```

chmorgan, 2019-03-18:
@Kiogora it looks like CONFIG_ESPHTTPD_SSL_SUPPORT is defined to 1, or enabled.

How are you seeing it not being available? Are you using it in your code and seeing compilation issues? If so can you provide your usage and the compiler errors here?

Kiogora, 2019-03-30:
Hi @chmorgan  I sm still looking into this but results are not yet forthcoming. I will have more time to dele deeper over the next 2 weeks.

closed 2020-10-30 by phatpaul