chmorgan/libesphttpd #67: Problem with SSL mode.

marbalon, 2019-06-17:
I have libesphttpd working find on ESP32 + esp-idf 3.3. Today I tried to enable SSL support. I got logs and everything seems to work fine but SSL_accept takes about 9 second...(see logs below). This is not the latest version, but I didn't see any commits related with this. My current version is 51084e0. About 140kb RAM free before I call https request.

Any ideas to solve this problem ?

`I (10618) httpd-freertos: address 0.0.0.0, port 443, maxConnections 16, mode ssl
I (10618) httpd-freertos: SSL server context create ......
I (10618) httpd-freertos: OK
I (10618) httpd-freertos: SSL server context setting ca certificate......
I (10628) httpd-freertos: OK
I (10628) httpd-freertos: SSL server context setting private key......
I (10958) httpd-freertos: init
I (11018) httpd-freertos: esphttpd: active and listening to connections on 0.0.0.0
D (11018) httpd-freertos: Sel add listen 54
I (11018) httpd-freertos: listening for new connections on '0.0.0.0'
D (13948) httpd-freertos: select ret
D (13948) httpd-freertos: SSL server create .....
D (13948) httpd-freertos: OK
D (13948) httpd-freertos: SSL server accept client .....
D (24388) httpd-freertos: OK
D (24388) httpd-freertos: Sel add listen 54
D (24498) httpd-freertos: select ret`

Thanks Marcin.

chmorgan, 2019-06-18:
Hi @marbalon what is the issue you are seeing? Is the listen failing?

marbalon, 2019-06-18:
The problem is long SSL_Accept - about 11seconds. Test above was made using wget command, but chrome restarts request every 5seconds.

D (13948) httpd-freertos: SSL server accept client ..... //waiting 11 seconds
D (24388) httpd-freertos: OK

chmorgan, 2019-06-18:
That’s pretty typical. The key exchange is a computationally expensive
process. I’m guessing we could specify our preference for faster key
approaches, if supported by mbedtls, to speed this up, but this kind of
time is what I’ve always seen on ssl connections on esp32.

Has it been quicker for you in the past?

Chris