chmorgan/libesphttpd #70: WebSocket link list crash on page reload

WayneKeenan, 2019-06-23:
Hi,

If I reload a page that has an active web socket connection then there is a crash at:
 https://github.com/chmorgan/libesphttpd/blob/master/util/cgiwebsocket.c#L149

Presumably the previous page's web socket is no longer valid and there should have been a LL update or guard here?

chmorgan, 2019-06-23:
Hi @WayneKeenan, stack trace? I'm guessing the same regarding the ll update but I'm not sure we know where to perform that yet.

WayneKeenan, 2019-06-23:
I didn't think it would help a great deal and there's a load of 'private' stuff, but then that was a bit presumptuous/looks good enough from my POV. 

Sorry to be lazy and not give more context, not my usual form.  
Thanks for the quick response.

edited:

```

chmorgan, 2019-06-23:
Hmm. Sure you didn't remove some libesphttpd stuff? Who is calling cgiWebsockBroadcast?

WayneKeenan, 2019-06-23:
all the blanked out lines are my code, the prior stack entry is:
```
0x4010c0d8: webserver_websocket_broadcast at .../my_project/src/components/webserver_helper/webserver_helper.c:135
```

WayneKeenan, 2019-06-23:
It just seems to be a case of calling a websocket update on a stale handle

chmorgan, 2019-06-24:
Doesn’t this mean your code is calling that function after the connection
was closed? I can’t recall the usage off hand but aren’t you receiving
callbacks when connections are terminated that would indicate not to send
via that connection? I could be totally off on the usage though.

I’m not defending the design or approach either, if we can improve it or
clarify we should do so.

Chris

WayneKeenan, 2019-06-25:
The API to publish a broadcast WS message is a NOP if no client has ever loaded the page.

Please look at your example.

But on reload the exact same call crashes.
There are no lifecycle management API’s, which is great. Just a link list bug...

chmorgan, 2019-06-26:
Hi @WayneKeenan. Hmm, yeah its simpler if we don't need any lifecycle management. I can't recall who has worked on the websockets most recently but I'm hoping someone will spot this and know what might be going on. If you have some idea about where the issue is we could look into it in more detail.

Si03, 2025-03-08:
The problem in my case was when calling cgiWebsockBroadcast frequently, eventually this will clash with a client connecting or disconnecting. the linked list gets adjusted while the broadcast is walking the linked list. boom: xTaskPriorityDisinherit tasks.c:5153 (pxTCB == pxCurrentTCBs[ xPortGetCoreID() ])
The solution is to put a semaphore around the three linked list users (xSemaphoreCreateMutex etc.)
I create the mutes on first call to cgiWebsocket(), and I ignore broadcast attempts until the sema exists.

phatpaul, 2025-04-01:
This old bug has just now become a major issue for us for no seemingly related reasons (perhaps just that our application has grown and is running many tasks and stressing the http server, especially during OTA upgrade.

It seems to be a problem with the lifecycle management of some memory objects which are referenced by 2 separate tasks/threads. The httpd task is responsible for creating the Websock objects, but then also adds them to a global linked-list which is used by cgiWebsockBroadcast() which could be called by other tasks. The objects are freed by only the httpd task in cgiWebSocketRecv() -\> websockFree() which also removes them from the linked list. The crash occurs when the cgiWebsockBroadcast() is walking the linked-list of Websock objects, but while walking, the list is modified and/or one of the objects is freed.

This bug is reproducible during our OTA flash operations: the flash erase/writes cause frequent CPU stalls and starve the task responsible for servicing cgiWebsockBroadcast(). When the browser detects a dead websocket connection, it closes it, which triggers cgiWebSocketRecv() to free it and remove it from the linked-list. But this often happens while cgiWebsockBroadcast() is still walking the list.

I had attempted a lame fix here: [22d227b](https://github.com/phatpaul/libesphttpd/commit/22d227b78ae990d71740f14aef006c384d7a6d0f) but it ironically just made the problem more prevalent.

I also tried adding a mutex to protect the linked-list from being modified while it was being walked, but this caused a dead-lock because the cgiWebsockBroadcast() needs the httpd task to do some work while it continues to hold the mutex locked. The httpd task needs the lock when a websocket is closed, so it blocks and dead-locks.

To reduce the complexity, I removed the linked-list and replaced it with just an array of pointers. That should be easier to manage if elements are added/removed to the list while it is being walked. The walking function will just skip over any pointers that are NULL.

Converting from linked-list to array didn't solve the issue, but will hopefully make it easier to debug. The memory overhead is pretty minimal, since the server only supports a finite number of connections/sockets anyway.

I added reference counting to protect lifetime of Websock objects using `kref.h` which was already in the library (used in cgi_wifi.c). Now references to the Websock objects are counted and only freed when the count reaches 0.

After all of this I have found it is still crashing in the same place: cgiWebsockBroadcast(): `if ((NULL != lw->conn) &&strcmp(lw->conn->url, resource) == 0) `I think what is happening is that the lw-\>conn object is being freed by the httpd task, but we don't clear our pointer to it.

Here's what I have so far: https://github.com/phatpaul/libesphttpd/tree/websocket_list_lifecycle
Any advice is appreciated.
I know there must be a better way to do this. I.e. why must we keep our own list of references to connections, when the httpd component already has (and owns the objects in) that list. And the httpd component should also already have a list of our Websock objects, since they are stored on `connData->cgiData`. Couldn't we just access that list which is (currently) internal to the httpd component?

chmorgan, 2025-04-02:
@phatpaul I'm afraid I've been away from this far too long to provide any detailed feedback. You've outlined the options available. It can be tough to remove an entry if the list is being iterated over in another thread, even if you can work the locking out. Sometimes using a refcount can help with cleanup there and you could lock and unlock around each list operation. If we could not have the cgiWebsockBroadcast() hold the lock that would remove the problem entirely, but again, I'm just rehashing what you've already mentioned.

phatpaul, 2025-04-03:
Thanks @chmorgan. 

I think I got if figured out. In addition to managing the life-cycle of the Websock objects, I needed to mark the ws->conn pointer as NULL when we disconnect. This allows the cgiWebsockBroadcast() to detect a closed conn and avoid trying to access a freed object.

I've hammered on it a bit with no more crashes. :)

PR #107 (2025-04-04): fix crash due to lifecycle of websocket list
Attempt to fix #70 

Seems fixed in my testing.
commit 57e0a0dc (2025-04-04): fix crash due to lifecycle of websocket list #107 #70

* Fixed #70 crash due to lifecycle of websocket list
closed 2025-04-04 by phatpaul