chmorgan/libesphttpd #83: ESP8266 RTOS SDK 3.3 Not Supported

mriksman, 2020-03-25:
`cgiflash.c` calls 	
`if (esp_image_verify(ESP_IMAGE_VERIFY_SILENT, &part_pos, &data) != ESP_OK) {`
line 614.
However, this function doesn't exist for this SDK. It exists as of 3.2 in ESP-IDF.

phatpaul, 2020-03-30:
Thanks for the report.  I admit I have never tested the esp8266 with the code that I contributed in that file.  

I wish we had a better/cleaner way to separate the core libesphttpd code from the utilities.  (i.e. maybe the utils are in a submodule, separated by target esp32/esp8266)
I also wish we had a CI setup to check that build succeeds for many different environments.

Anyway, please submit any fixes as PR and I will merge as long as they don't break my build for ESP32.

phatpaul, 2020-10-31:
I added a disclaimer to the readme that ESP8266 not supported on this fork.

Hopefully someone will join us to maintain ESP8266 support...

closed 2020-10-31 by phatpaul