chmorgan/libesphttpd #84: WebSocket platLock

NikLeberg, 2020-03-30:
Hi,

I've ran into some bugs while using the cgiWebsocket implementation in a project on an esp32:
- Error "send buf tried to write %d bytes, wrote %d" in [httpd.c](https://github.com/chmorgan/libesphttpd/blob/f7ac4636b915ceb4e82b14ae079a45be63ec9866/core/httpd.c#L448) got triggered sometimes. But allways with more bytes written than it said it should have (e.g. "tried to write 0 bytes, wrote 19").
- Some messages from server to client were sent twice.
- Frameheaders got corrupted (Chrome throws errors and closes connection). Some could not be decoded as valid utf-8 and some had reserved bits turned on.

A recording with wireshark revealed that in such cases, all bytes get sent sort of "correctly" but out of sync (e.g. in the TCP-Stream the WebSocked-Header should start but some random other bytes were sent before the correct header is finally sent). This lead to the wrong decoding in the browser as the header was not where it should have been.

So I've dug inside the code and noticed that inside all cgi-functions the sendBuff is guarded with 
[`httpdPlatLock()`](https://github.com/chmorgan/libesphttpd/blob/f7ac4636b915ceb4e82b14ae079a45be63ec9866/core/httpd-platform.h#L14) and [`httpdPlatUnlock()`](https://github.com/chmorgan/libesphttpd/blob/f7ac4636b915ceb4e82b14ae079a45be63ec9866/core/httpd-platform.h#L15). This works great if the server only ever reacts to requests. But with a WebSocket the server is allowed to send at any time. With functions [`cgiWebsocketSend()`](https://github.com/chmorgan/libesphttpd/blob/f7ac4636b915ceb4e82b14ae079a45be63ec9866/util/cgiwebsocket.c#L113) and [`cgiWebsocketClose()`](https://github.com/chmorgan/libesphttpd/blob/f7ac4636b915ceb4e82b14ae079a45be63ec9866/util/cgiwebsocket.c#L155) the server sends data to the client without a request and the httpd instance and sendBuff is unguarded.

I've added the locks and tested extensively. With the following code none of the mentioned bugs appeared:
```c
int ICACHE_FLASH_ATTR cgiWebsocketSend(HttpdInstance *pInstance, Websock *ws, const char *data, int len, int flags) {
	int r=0;
	int fl=0;

	// Continuation frame has opcode 0
	if (!(flags&WEBSOCK_FLAG_CONT)) {
		if (flags & WEBSOCK_FLAG_BIN)
			fl = OPCODE_BINARY;
		else
			fl = OPCODE_TEXT;
	}
	// add FIN to last frame
	if (!(flags&WEBSOCK_FLAG_MORE)) fl|=FLAG_FIN;

    if (ws->conn->isConnectionClosed) {
        ESP_LOGE(TAG, "Websocket closed, cannot send");
        return WEBSOCK_CLOSED;
    }

	httpdPlatLock(pInstance);
	sendFrameHead(ws, fl, len);
	if (len!=0) r=httpdSend(ws->conn, data, len);
	httpdFlushSendBuffer(pInstance, ws->conn);
	httpdPlatUnlock(pInstance);
	return r;
}

void ICACHE_FLASH_ATTR cgiWebsocketClose(HttpdInstance *pInstance, Websock *ws, int reason) {
	char rs[2]={reason>>8, reason&0xff};
	httpdPlatLock(pInstance);
	sendFrameHead(ws, FLAG_FIN|OPCODE_CLOSE, 2);
	httpdSend(ws->conn, rs, 2);
	ws->priv->closedHere=1;
	httpdFlushSendBuffer(pInstance, ws->conn);
	httpdPlatUnlock(pInstance);
}
```
Only problem now is, that the lock possibly blocks the calling task without a timeout. In my usecase no problem, but I don't know if it is the best solution.

I hope this helps as I find this lib really great and I want to give a little bit back.

Cheers
Nik

phatpaul, 2020-03-31:
Thanks for contributing!  This looks like a quality improvement.
I'm happy to test it on the example project (which just sends timestamps and echos messages).  

If the lock is a problem for someone, they could just implement a queue between the cgiWebsocketSend and their task.  Right?

NikLeberg, 2020-03-31:
Yeah, that's how I implemented it actually.

One quick heads up: Forgot that one also has to add the header `#include "httpd-platform.h"` inside [cgiwebsocket.h](https://github.com/chmorgan/libesphttpd/blob/ea677d587c53cd2cbedbad41ea8231ad4ddb27d4/include/libesphttpd/cgiwebsocket.h#L4).

phatpaul, 2020-10-30:
Please send a PR @meowthink

meowthink, 2020-10-31:
> Please send a PR @meowthink

Would @NikLeberg himself like to do so? I definitely would help by author's will.

NikLeberg, 2020-10-31:
I just did! It's my first PR so please tell me if something is off.
My project is a bit screwed up: I have to "flatten" the folder structure of my in-use-copy of libesphttpd. Please confirm that the header "httpd-platform.h" can be found when building.

closed 2021-04-29 by phatpaul