chmorgan/libesphttpd #90: Malformed encoding in chunked resposes

errno, 2020-10-27:
I was writing a CGI function which generates quite big responses (>2k). So it utilizes consecutive `httpdSend()` calls and `HTTPD_CGI_MORE` return code.
At some point I've stumbled upon corrupted responses. `curl` says `Malformed encoding found in chunked-encoding`.

Further digging revealed that I managed to fill `conn->priv.sendBuff` to the brim (exactly 2048 bytes) before returning from the function with `HTTPD_CGI_MORE`. That, in turn, caused a problem in `httpdFlushSendBuffer()`, because there is no room for closing a chunk with `\r\n`, and chunk length is calculated incorrectly.

I'm going to try to fix this issue myself. If I succeed I'll open a pull request.

errno, 2020-10-27:
That was relatively easy. I hope my approach is acceptable. And my changes didn't break other stuff :)

phatpaul, 2021-04-29:
Merged (better late than never right?).  Thanks.
BTW, maybe https://github.com/chmorgan/libesphttpd/issues/93 is related to this issue.  I'll test that again now that this is merged.

closed 2021-04-29 by phatpaul

errno, 2021-04-29:
Hurray! It's just the right time.
I've returned to working on the project in which I discovered this issue a few days ago, so now I can switch back from my fork :)