chmorgan/libesphttpd #92: Static files handling

yzazik, 2020-10-30:
Hello, (sigh) well, this is not really an issue any more, or maybe not yet again since I tried to use this code to give my web server a bit of asynchronicity but using it turned out to be more problematic than I expected so I left this idea alone for now, but.
I thought that the idea behind serving static files should have been in... well, not serving them as dynamic ones rather than letting them to be "cached forever" (like that was ever any good). By that I mean, rather send (initial) response with E-Tag or Last-Modified with max-age let's say 2-5min (I'd rather send "304 Not Modified" more often than let the file be cached for a year and keep pressing F5 on every page I switch to) just enough time till I re-upload a new image :-). And then there is no need in static file distinction since with this strategy dynamic content can be served the same way as long as dynamic stuff hasn't expired.
E-Tag is some sort of CRC/MD5 and rather identifies the file content rather than the file properties (name, location, mtime, etc). Kinda heavy to evaluate on the fly and storing it with the file would mean that ESPFS would need either extra flag and field to save it after the header or trail after the file content similar to NTFS streams which I guess a bit cleaner way of having filename.md5 files around.
Last-Modified requires just file MTIME (modified time) which is, guess what, not in ESPFS either. It's a shame really, reserving 256 bytes for a file name using only like a third of it and not sacrificing 4 bytes for time_t. One could use app image compile time instead but that only works if ESPFS is linked with the image. And again, adding more fields in the header without breaking backward compatibility would probably require as much code as rather using dynamic headers (like TLV encoding).
And couldn't really send 304 Not Modified since everything is send with "OK" (304 OK in this case) and buried deep down, more copy-paste-modify code, making original one... well, left out.
Someone really have to remove ESP8266 tag from the page, this code won't build on v3.3 8266 SDK. Mostly because... yep, guessed right, ESPFS, no similar mmap, no pyhton3, no heatshrink2 that currently works there.

phatpaul, 2020-10-30:
That was a bit hard to follow, but I think you're wanting to optimize the browser's caching of static files.  I have worked on that for some time and have a solution that works well for me.

1.  index.html is always served with the header: `Cache-Control: no-store, must-revalidate, no-cache, max-age=0` so that it is never cached by a browser (a well-behaved browser).  Maybe not ideal, but it's only a few kB.
2.  my static content (other than index.html) is served from a subdir /static/ and httpd is configured to serve files within with: `Cache-Control: max-age=365000000, public, immutable`.  That tells the browser that it can cache those files forever.  The only way I can ensure that the browser refreshes those files is to rename them (or add some cache-busting query with a random number which is ignored by httpd i.e. `http://myserv/static/staticfile.js?cache-buster=23552143423` )
3.  my static content is generated with a build script (gulp, google-closure-compiler, npm, and many gulp plugins) which adds a hash to the filename of each file, generated from the file's content.  i.e. `static/app-e61e3967ff.min.css`
4.  The build script automatically modifies the <script> and css links in index.html to point to the generated ones in /static/.  i.e. 
```
<html><head>
<link rel="stylesheet" href="static/app-e61e3967ff.min.css">
<script src="static/app-e9c87a5886.min.js"></script>
```

So, if no static content has changed, the browser only has to get index.html (a few kB).  But if I have rebuilt my webapp, then the browser must also get all of the changed static content (and only the content that actually changed).

See my comments in https://github.com/chmorgan/libesphttpd/pull/75 for how to set up httpd to send different headers per file/folder.

Would that work for you?  I can provide my build script if you want.

You also said `Someone really have to remove ESP8266 tag from the page, this code won't build on v3.3 8266 SDK. `
I agree 100%.  I have never used ESP8266 and I have surely added several breaking changes.  We are just working on our own ESP32 projects and contributing what we think is useful back here without testing on ESP8266. 
Someone should fork this (a previous working version) for ESP8266 and maintain it separately.
@chmorgan can we add a note to the main readme that this is not expected to work with ESP8266 to avoid people wasting their time?

chmorgan, 2020-10-30:
@phatpaul totally on board with mentioning that no one is actively maintaining esp8266 support. I'm surprised that the 8266 has gone as long as it has given the cost and how long the esp32 has been out but yeah, without someone fixing it its tough.

phatpaul, 2020-10-31:
added warning to the readme.

closed 2020-10-31 by phatpaul

yzazik, 2020-10-31:
@phatpaul Thanks for prompt reply. Well, yes, I understand the idea with Cache-Control.
If you are not familiar with conditional requests the highlights of the flow are:
1. Client requests the resource through regular GET.
2. Server responds with 200 OK + Last-Modified header with the time when the content was last changed + content (body) itself.
3. Client caches the content and remembers it's modification time for the duration of the content age (age is server controlled and there is some kind of default if not provided, let's say an hour), during this time (till age expires) client won't request the content from the server at all (unless forced), that's it, even if the content changes before age expires client wouldn't know and won't send requests for it, just gets from local cache (that is where the most of user issues come out).
4. After age expires, client would send a conditional GET request with If-Modified-Since time (the time that server sent with the last response).
5. Server responds with 304 Not Modified w/o content (header only, very short) if content (in this case file modification time) hasn't changed since the time in If-Modified-Since OR if content has indeed changed it responds with 200 OK + new Last-Modified time + new content.
This is quite convenient since apart from file's modified time there is no extra build/script/makefile effort is needed, in any case, this is the one I was missing in ESPFS. Might still be worth introducing mtime to esp32-espfs (others (spiffs, littlefs) do have it). I made the change locally to serve my purpose but I just added it to the header breaking compatibility, but it is possible to do it properly.
The other problem is that serving static files from specific directory (/static/) requires rearranging files from existent well established web designs my client has (stubborn too), rather doing it by file extension (similar to ESPFS processing) would be more convenient.
Serving non-static content with templates is also possible with conditional requests, means, conditional responses for template requests could also be 304 if variables' values (used in template substitutions) haven't changed, well, this would still require holding a timestamp in the CGI handler, but this would be a custom user handler. Also, when the file is being served, there are no hooks for responding with 304 code, needs redesign.
I'm reflashing ESPFS partition with HTML content more often (like few times a week: all sorts of customizations per device/client/environemnt/etc) and application image maybe once in a few month (at least till things will settle down), and I want browser only cache files for a short time and force it to make conditional requests (with If-Modified-Since) rather than not making any requests at all (which what is happening with "long" Cache-Control ones).
E-Tag is another similar way of doing it but it usually requires CPU cycles to actually check the content and produce the crc/hash (E-Tag), this is where it has a similarity with your approach, but you are saving it as part of the file name rather than somewhere in the file system but attached to the file (like in EspFsFile or as I said some sort of NTFS stream) and changing filenames is a maintenance nightmare in my case, just HTML code and source control tracking for .css/.html/.js would be a project by itself.
But, as I said, no worries, I found my ways with IDF's esp_http_server, due to a tight schedule, but I might give it another try later.
No offense, but I was wondering if someone tried to port lighttpd to ESP32, I bet memory requirements are off the scale but might be simply excessive due to a fear of memory deficit :-). If not, well, maybe I do need to get my hands dirty once I finish with my current project.
ESP8266 is still out there, could be smaller and cheaper, I only used it on sensors and relays and only because I couldn't find ESP32 boards with battery charger at a time, but usually all you need is a few pages for status, config and administration and maybe a couple of websockets, still, too bad IDF is single response driven.
Thanks again.